keycard-tech/keycard-shell #20: Implement e-label

bitgamma, 2025-04-09:
details tbd

guylouis, 2025-07-13:
**Context** 
Products sold in europe and in the usa need a list of mandatory mentions on the product like
CE mark 
barred bin 
model number
name of the manufacturer 
power rating 
serial or lot number 

Products who have a display can use the display to show these informations instead of printing them on the product. 

**For Shell**
On Shell we will have the CE mark and trashed bin. Both on the PCB behind the battery. But not the rest of theinformations. 

So we have to create a menu, accessible when the product is unlocked. 

**To be confirmed (I will be able to conifrm week of 28/7/2025)**
- place and name for the menu (probably will be something like settings/device/legal) I have to check what's the tolerated maximum number of clicks to get to the menu 
- content of the screen. Hopefully it will be only text (since CE and barred bin are already on the product), and the only variable one will be the SN of the device

guylouis, 2025-07-30:
**Regarding where to put this elabel, the reference docs are** 

https://fcc.report/FCC-ID/A4RGUIK2/5042986.pdf

[image]

So I suggest to put it here: settings/device/**regulatory label**

**Regarding the content of the label it should be**

"
Brand: Keycard 
Model: Shell 
Input: 5V DC, 1A
Manufacturer: Status Research and Development Am Zirkus 2, D-10117 Berlin, Germany
Made in China 
Serial number:xxxxxxx
"

note: actually I think the input could take up to 1.5A but we can play it safe and leave 1A here

bitgamma, 2025-08-11:
done