keycard-tech/keycard-shell #21: setup reproducible builds

bitgamma, 2025-04-09:
the only thing missing is fixing the compiler version and either doing a script or writing instructions to recreate the build env

bitgamma, 2025-06-04:
tested my own build instructions from the [README](https://github.com/keycard-tech/keycard-shell/blob/master/README.md#building-the-firmware) on 2 computers with different OSes and the build was reproducible as expected. If anyone else wants to test it - it'd be good

The only open question is if the documentation is clear enough (consider this is oriented to developers).

cc @guylouis @cheny0

guylouis, 2025-06-04:
when we provide the builds on github (we will do because we want users to be able to upload themselves the firmware with https://github.com/choppu/kpro-updater, shouldn't we provide the public part of the signing key we're using (at this stage, and when we go in prod we will update it with the update value, if it differs) if someone wants to compare the result of his build with the builds we provide? 

but at this stage, it looks good to me 

we can make the doc better if we get feedback from anyone in the community about it needing help

guylouis, 2025-06-04:
great, we can close this one then

closed 2025-06-04 by guylouis