keycard-tech/keycard-shell #23: support SeedQR and text-mnemonic QR

bitgamma, 2025-04-09:
add support for https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/README.md and/or plaintext mnemonic QR.

bitgamma, 2025-06-25:
proposed (and currently implemented) interaction. The default choice is "with passphrase" since it seems to be very common to set one when using SeedQR.

Please review @xAlisher @guylouis:

[image]

[image]

[image]

[image]

xAlisher, 2025-06-26:
@guylouis @bitgamma

Following up on the conversation here:
[https://discord.com/channels/1366754869561790474/1387350462117712045/1387690470423007333](https://discord.com/channels/1366754869561790474/1387350462117712045/1387690470423007333)

I’m aligned with @bitgamma on the minor copy updates.

User flows have been added to Figma:
[https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=5295-93739\&t=PKWxt5CPmQwZbo0K-1](https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=5295-93739&t=PKWxt5CPmQwZbo0K-1)

Great job on this feature! 👏