keycard-tech/keycard-shell #29: Update auto-complete interaction and feedback design

xAlisher, 2025-05-14:
### Title
Update auto-complete interaction and feedback design

---

### Description
The current auto-complete input flow is not intuitive for many testers.

**Reported by:** 9 testers

#### Key issues:
- It’s unclear what happens when users hold to save a word — most testers only understood it after experimenting.
- Selecting a suggested word feels confusing; several testers tried using arrow keys to confirm.
- Typing is slow and lacks visible feedback for how suggestions are selected or confirmed.

---

### Suggested improvements:
- Improve on-screen labels and titles for clarity
- Display a visible list of suggested words
- Add a walkthrough video or animation to onboarding materials
- Show a validation error (e.g. red text) if a user tries to confirm an invalid word

### Resources
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

xAlisher, 2025-05-24:
Plan Minimum
- Update wording to make expected behavior explicit.
- Improve readability of titles and the highlighted word segment.
- Clarify action label to reinforce what’s being confirmed.
- Increase space between cursor and word parts for readability

[image]

[image]

Plan Maximum

**Introduce an alternate suggestion input:**
- Show a full list of valid words to choose from.
- Reduce number active keys with every character selected to minimize navigation steps.

Overview
This update targets common tester friction in the current flow. Plan minimum is low-effort and already resolves major confusion. Plan maximum adds predictability and significantly faster word input.

Proposal: https://www.notion.so/Shell-input-improvements-14d8f96fb65c80f0bf76ff1ff1190570  (note: design and keyboard layout outdated)
Figma https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=5055-140967&t=5ZdeBNICwB1bczF1-1

🙏👁️ @guylouis @bitgamma

xAlisher, 2025-05-24:
Error handling: 
- Remove "Hold to" button

[image]

bitgamma, 2025-05-29:
extracted in #111  the more complete autocomplete solution

bitgamma, 2025-05-29:
implemented the minimum solution, however the "Hold to confirm label" actually doesn't fit in screen so I left the current label for the action label

bitgamma, 2025-06-04:
In code the space between navigation icon and screen border is 16px. The space between icon and label is 10px. I can decrease them if needed

closed 2025-06-23 by bitgamma