keycard-tech/keycard-shell #39: Make battery easier to remove through design adjustment

xAlisher, 2025-05-14:
### Description
Several testers found it physically difficult to remove the battery from the device. The current design does not provide enough leverage or grip to extract it comfortably.

**Reported by:** 3 testers

#### Key issues:
- Battery is tightly fitted and lacks a convenient way to remove it.
- No designated area to grip or push the battery out.
- Frustration during basic maintenance impacts perceived usability.

---

### Suggested improvements:
- Add a notch, groove, or indentation to assist with finger grip.
- Consider introducing a pull tab or tool-free removal mechanism.
- Ensure that battery maintenance feels simple and intentional.

xAlisher, 2025-05-19:
It's possible to remove the battery with the card. 
Add this to the doc.

guylouis, 2025-05-23:
conclusion is indeed that we won't make changes to the plastic but document precisely how to remove the battery