keycard-tech/keycard-shell #43: Adjust long-press timing to prevent accidental triggers

xAlisher, 2025-05-14:
### Description
Several testers noted that the current long-press duration is too short, leading to accidental activation of long-press actions when they intended a regular press.

**Reported by:** 2 testers

#### Key issues:
- Long-press is triggered too quickly.
- Users unintentionally activate long-press actions during normal use.
- Causes confusion or unexpected behavior, especially during input

---

### Suggested improvements:
- Slightly increase the duration required to trigger a long-press action.
- Add simple visual feedback (e.g. animation) to reinforce that a long-press is in progress.

---

### Resources  
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

xAlisher, 2025-05-19:
time x3

bitgamma, 2025-05-28:
cc @guylouis @xAlisher

bitgamma, 2025-06-03:
to me even 200ms (current time seems too long). Wdyt? @xAlisher @guylouis