keycard-tech/keycard-shell #47: Improve packaging usability — box is difficult to open and remove Shell from

xAlisher, 2025-05-14:
### Description
Testers reported difficulty both opening the Shell packaging and removing the device from the box. These friction points negatively impacted the unboxing experience and first impressions of the product.

---

#### Key issues:
- Outer box lacks intuitive opening mechanism; users had to guess or apply force.
- The Shell device is tightly fitted with no clear way to lift it out.
- Risk of user frustration or device damage during unboxing.

---

### Suggested improvements:
- Redesign the box for easier opening (e.g. tear strip, pull tab, or clear indicators).
- Add a finger recess, ribbon, or lift tab to help remove the Shell device easily.
- Prototype revised packaging for smoother, one-handed unboxing experience.

guylouis, 2025-05-14:
working on providing an improved version
- same construction philosophy
- with more room inside, will facilitate grabbing stuff 
- with higher quality feeling of the material 

1st step is to decide if we can keep the black & orange type of design and improve the perceived quality. It's unsure while using corrugated paper that we can manage too, because of the edges 

If we can't we'll have cardboard color material on which we will print our design. It will help have a cleaner looking edge.

guylouis, 2025-05-23:
strategy is to create a place to put the finger to lift up Shell and the papers

[KEYCARD_coffret_v1_23052025.pdf](https://github.com/user-attachments/files/20414222/KEYCARD_coffret_v1_23052025.pdf)

closed 2025-06-30 by xAlisher

guylouis, 2025-07-14:
packaging had been improved, see the die cut below 
samples to be received end of July - early August max

[image]