keycard-tech/keycard-shell #49: Improve readability of word prompt during backup flow

xAlisher, 2025-05-14:
### Description
During the backup process, users found it unclear which word they were expected to input. The current title bar, which displays the word number, is not easily readable due to low contrast.

---

#### Key issues:
- Word number (e.g. “Word 5”) is hard to see against the background.
- The label lacks contrast and visibility, leading to hesitation and confusion.
- Users are unsure which word they’re confirming or typing.

---

### Suggested improvements:
- Increase the contrast of the title bar
- Consider moving the word number to a more visible location, such as below the title
- Explore adding visual cues (like progress dots or highlighting) to reinforce the step number.

---

### Resources  
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

xAlisher, 2025-05-24:
Suggest improving visibility without disrupting layout by:

- Increasing the title font size (may create slight tension in other areas, but it’s manageable).
- Boosting contrast between title text and background by brightening the style.

[image]

Bonus idea:
- To improve predictability, we could display the number of remaining steps to complete the flow.

[image]

🙏👁️ @bitgamma @guylouis

bitgamma, 2025-05-28:
@xAlisher lgtm, but I see in the figma design that some title needs to be changed because they don't fit with the larger font size.

Also, I see highlighted words/menus are now using white font instead of black. Is this change part of this issue?