keycard-tech/keycard-shell #50: Clarify behavior of Cancel button on word input screen

xAlisher, 2025-05-14:
### Description
Testers were unsure what would happen if they pressed the “Cancel” button during the word input screen in the backup flow. The lack of explanation caused hesitation and fear of losing progress.

---

#### Key issues:
- “Cancel” button lacks contextual clarity — does it exit the flow, go back, or discard progress?
- Uncertainty at this step creates friction and user anxiety.
- No confirmation or explanation is provided when the button is pressed.

---

### Suggested improvements:
- Add a short label or tooltip to clarify what “Cancel” will do.
- Consider prompting a confirmation modal if pressing “Cancel” would result in data loss or reset.
- Use more descriptive text (e.g., “Go back” or “Exit backup”) if appropriate.

---

### Resources  
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

phift, 2025-05-29:
Maybe that's what you mean by backup flow. But in the `Import recovery phrase` screen, 'c hold to exit' is similarly confusing, because holding c doesn't 'exit' the flow, but it returns to the previous screen

bitgamma, 2025-07-14:
labels have been adjusted accordingly

closed 2025-07-14 by bitgamma