keycard-tech/keycard-shell #56: Chevron-right inside the circle icon misinterpreted as  chevron on keypad

xAlisher, 2025-05-14:
### Description
Testers confused the chevron-right icon inside an orange circle with the physical chevron key on the keypad. This led to hesitation and incorrect inputs. The blending of icon roles reduced clarity in the interface.

#### Key issues:
- Chevron inside the orange circle is visually similar to the keypad's left-arrow button.
- Users mistakenly interpret it as a hardware control rather than an on-screen indicator.
- Inconsistent icon behavior breaks visual expectations.

---

### Suggested improvements:
- Consider using a plain orange circle for confirmation actions (OK), without overlaying the chevron.
- Keep all chevron icons on-screen consistently white.
- Ensure visual roles between hardware keys and on-screen symbols are clearly separated.

---

### Resources  
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

xAlisher, 2025-05-23:
OK button states simplified and unified. Now we'll have only two states. Default and long press. 

[image]

[image]

[image]

I need one more pair of eyes for feedback and review.
@guylouis @bitgamma

guylouis, 2025-05-23:
looks good! 
maybe write it somewhere in the doc in a 'navigation' part of the faq

bitgamma, 2025-05-23:
looks good to me too

bitgamma, 2025-05-28:
done

closed 2025-05-28 by bitgamma