keycard-tech/keycard-shell #58: Allow adding a password to mnemonic

bitgamma, 2025-05-14:
often called the 13th or 25th word, we could allow the user to add a password the mnemonic phrase. This is especially useful if we implement #23 since it is relatively common to use a password in this case.

This step would happen during onboarding of a new card when:

- importing an existing mnemonic (either keyboard or QR)
- generating a new mnemonic, after backup and before the keys are actually loaded

bitgamma, 2025-06-09:
@xAlisher 

my understanding is that in both generate mnemonic and import mnemonic the menu where you select the mnemonic length will become something like:

12 words
12 words + pass
24 words
24 words + pass

and that after the mnemonic has been entered or backed up the user will enter the pass.

Is this also how you see the flow? If that's the case I will start implementing this.

bitgamma, 2025-06-10:
Implemented the feature so that it at least works. For UI I simply modified the mnemonic length selection menu and added, in case an option with passphrase has ben selected, a final screen to input the passphrase. Feel free to revise this in any way

[image]
[image]

xAlisher, 2025-06-23:
@bitgamma design updated

Import recovery phrase https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=5235-62008&t=4bPH3kMTDniDkBOV-1
Generate keys
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=214-58571&t=4bPH3kMTDniDkBOV-1

closed 2025-06-23 by bitgamma