keycard-tech/keycard-shell #62: Clarify wallet type selection (Ethereum, Bitcoin, Multichain)

xAlisher, 2025-05-14:
### Description
Testers were confused when asked to choose between Ethereum, Bitcoin, or Multichain wallets during setup. The options lacked context or explanation, making it unclear which one to pick.

---

#### Key issues:
- Users didn’t understand the differences between Ethereum, Bitcoin, and Multichain wallets.
- Lack of in-device guidance made this decision feel risky or arbitrary.
- No clear link to supported wallet features or compatibility.

---

### Suggested improvements:
- Add a brief description under each wallet type (e.g. “Ethereum: EVM-compatible chains”).
- Consider adding recognizable brand names (e.g. MetaMask, Rabby, OKX) to improve comprehension.
- Update documentation to explain each wallet type, use case, and compatibility in more detail.

---

### Resources  
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

xAlisher, 2025-06-03:
- remove Sparrow
- remove multichain

xAlisher, 2025-06-23:
[image]

https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1&t=4bPH3kMTDniDkBOV-1

I'm suggesting to name most popular wallets right in the menu for better user experience. 

@guylouis @bitgamma WDYT? 

@cheny0 could you give me the list of wallets compatible with Shell sorted by popularity to add to this menu, please?

xAlisher, 2025-06-23:
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=53-19251&t=4bPH3kMTDniDkBOV-1

cheny0, 2025-06-25:
@xAlisher Here's a list in the current docs: https://keycard.tech/docs/supported-wallets#wallets-for-keycard. The "Multichain" QR code works for Bitget and backpack

guylouis, 2025-06-30:
I wouldn't trust this list too much, it was more a marketing list than a result of good testing 

here's where we will track the result of the real testing 
https://github.com/keycard-tech/keycard-shell/issues/81

guylouis, 2025-08-13:
So the entries will be:
Ethereum 
Bitcoin 
Bitcoin Multisig 
Bitget

Bitcoin Testnet is present for now but will be removed before launch

guylouis, 2025-09-08:
I am unclear if it's ok like this (current implementation is ethereum, bitcoin, bitcoin multisig, bitget) or if we want to add more details in the list

closed 2025-09-08 by xAlisher