keycard-tech/keycard-shell #67: message when entering card with no chip

guylouis, 2025-05-14:
if someone enters his card backwards (chip on the bottom) then the product doesn't turn on, can we turn on and display a message like "is your card correctly inserted? the chipset should be on top side"

bitgamma, 2025-05-14:
There is a generic disconnection message, but a more precise one can be given in theory. The problem you see with no turning on is just a longer than needed timeout that I fixed yesterday, now it only waits 1sec

guylouis, 2025-05-19:
an indication like 'chip needs to be upfront' or even a picture would make a lot of sense since users putting the card upside down might be a very common problem 

@xAlisher any ideas on text or illustration ?

closed 2025-05-19 by xAlisher