keycard-tech/keycard-shell #70: Improve error message for unsupported Bitcoin Taproot transactions

xAlisher, 2025-05-14:
### Description
When attempting to sign a Taproot-based Bitcoin transaction using Shell and Unisat (e.g. for buying an ordinal on Magic Eden), the device displayed a misleading “wrong key pair” error. This happened even though the same Keycard worked for connecting to Unisat. The actual issue is related to Taproot being unsupported.

#### Context:
- Taproot is currently not supported.
- The transaction included Taproot components, leading to a signing failure.
- Error message displayed: “wrong key pair,” which does not explain the real cause.

---

### Suggested improvements:
- Update the error message to clarify that Taproot-based transactions are not supported.
- Suggested phrasing: _“This transaction uses a Taproot address, which is not supported yet.”_
- Optionally include a link to documentation or help page for supported address types.

---

### Notes:
Tested with:
- Shell + Unisat
- Magic Eden / Ordinal TX
- Switching to Segwit resolved the issue

---

### Resources  
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

xAlisher, 2025-05-21:
> just update wording to a more generic error message to say that there is nothing we can sign here (it's not a question of keys)

But also could be a wrong key pair. 
@guylouis @bitgamma WDYT?

[image]

guylouis, 2025-05-22:
seems good to me
how can we track all the things that need to be in the documentation ? 
this is one for instance
any idea @cheny0 ?

cheny0, 2025-05-23:
@guylouis, what we are doing now seems to be good enough, which is to add the "documentation" label to relevant issues. When you want to check all documentation-related issues, you go to the project board and click on the "documentation" label. Then you'll see only issues with this label, like this https://github.com/orgs/keycard-tech/projects/1/views/1?filterQuery=label%3Adocumentation. WDYT?

guylouis, 2025-05-23:
great to hear!

bitgamma, 2025-05-28:
the subtext message is too long, otherwise I have to rework the entire info dialog to measure all text and calculate the vertical offsets before drawing, which I don't think it is worth. Let's omit the "Visit ..." part

closed 2025-06-03 by bitgamma