keycard-tech/keycard-shell #73: improve old applet message

guylouis, 2025-05-19:
# situation 

when a user uses an old keycard applet (see definition below) the following message shown is 

[image]

@bitgamma could you list which criterias are considerded to state a card has an old applet 

# proposition 

add a link (QR code) to a github page on how to update the applet

bitgamma, 2025-07-21:
@guylouis is this still relevant? I didn't get what we should do

guylouis, 2025-07-22:
This is the rare case where one will use a keycard v1: it uses keycard UID, hasn't got the certificate, and doesn't work with Shell since it doesn't export the extended public key for Shell to register with a software wallet. 

To make it simple, can you just change the text to 

'Old applet detected'
Your keycard is too old to work with Shell. Please reach to our support team for help.

bitgamma, 2025-08-06:
done