keycard-tech/keycard-shell #76: Update Shell packaging design

Francesca-G, 2025-05-19:
### Description

Update Shell packaging design

- **COLORS →** Natural cardboard color + orange 🟠
- **FLAPS →** need to be bigger and black (useful to protect the device)
- **GLOSS PRINT →** thicker lines, consider embossed gloss print to make it more visible
- **INSTRUCTION CARD →** Change QR code to lead to a START page for Shell (/start/shell)
- **CARD HOLDER →** change to a single piece of paper with 4 slits and a card on each side (no folds) - each card uses 2 of the slits on each side
- **INSIDE DIVIDER FOLD →** make the hole slightly bigger to make it easier to grab

-------------------------------------------------

### Order of things in the box ⬆️

1. DEVICE
2. INSTRUCTION CARD (STEP NUMBERS FACING UP)
3. KEYCARDS
*—————— FOLD DIVIDER ——————*
4. INSTRUCTION & SAFETY USER MANUAL
5. BATTERY + CABLE

guylouis, 2025-05-19:
we will need to reserve a place on the packaging to stick a sticker with the serial number 
https://github.com/keycard-tech/keycard-shell/issues/80

guylouis, 2025-05-23:
[KEYCARD_coffret_v1_23052025.pdf](https://github.com/user-attachments/files/20412076/KEYCARD_coffret_v1_23052025.pdf) new prototype

guylouis, 2025-07-14:
packaging had been improved, see the die cut below
samples to be received end of July - early August max

[image]