keycard-tech/keycard-shell #81: create a file with interop tests between shell and software wallets

guylouis, 2025-05-20:
let's create a list of software wallets that use QR both in EVM world and Bitcoin and test Shell against them 

the file should be accessible from anyone

to be decided where we post this? github would be great but probably an accessible notion page would be easier to read and comment by any one

guylouis, 2025-06-30:
started a doc to structure the list of tests 
https://docs.google.com/spreadsheets/d/1EiACz-Knydb4PJf9S4bA0mm3f5CatSbe6x87P2xm38g/edit?gid=0#gid=0

romanzac, 2025-07-24:
Related issue created https://github.com/keycard-tech/keycard-shell/issues/145

romanzac, 2025-08-05:
@xAlisher @guylouis Detailed test result available at [Notion doc](https://www.notion.so/Keycard-Shell-Test-Results-22b8f96fb65c80faa06fcc195373624b?source=copy_link)

Based on test results, Shell currently works well with:

### Ethereum
- Metamask
- Rabby 
- Imtoken (note: not btc support)

### Bitcoin
- Sparrow 
- Specter 
- Nunchuk
- Unisat

Multichain 
- Bitget (eth + btc)

guylouis, 2025-09-08:
update of the list of compatible wallets (08/09/2025)

**Ethereum**
Metamask
Rabby
Imtoken (note: not btc support)
**Bitcoin**
Sparrow
Specter
Nunchuk
Unisat
Bitcoin Safe
**Multichain**
Bitget (eth + btc)

closed 2025-09-08 by guylouis