keycard-tech/keycard-shell #82: test bitcoin multisig

guylouis, 2025-05-20:


bitgamma, 2025-06-04:
currently popular MultiSig schemes for bitcoin need to be tested. Needs some research to understand what is currently relevant. @phift can you help with this?

phift, 2025-06-05:
## Possible implementation

Generation | On-chain format | Crypto primitive | What Keycard still needs | Difficulty on Keycard | Roll-out/Relevance
-- | -- | -- | -- | -- | --
1 | P2SH (legacy “3…”) | ECDSA CHECKMULTISIG | Nothing new cryptographically; just add P2SH redeem-script handling in the MCU parser. | Low – the secure element already does ECDSA; extra code is small. | Probably not needed
2 | P2WSH (native SegWit “bc1q…0…”) | ECDSA CHECKMULTISIG | Nothing. PSBT-v0 and current ECDSA engine are sufficient. | Low – ready today. | covers the vast majority of live multisig flows.
3a | P2TR script-path | Schnorr CHECKSIGADD | BIP-340 Schnorr signing + x-only pubkey export. | Medium – fits in JCOP4; ≈ 40 ms per sig. | Users keep the same MCU firmware; flash a new Java-Card applet
3b | P2TR key-path | MuSig2 (n-of-n key aggregation) | Schnorr + 2-round nonce storage per input + new PSBT keys. | Hard – needs persistent nonce slots and anti-reuse logic inside the card. | Could become relevant for a Shell v2 – once BIP 327 is final
3c | P2TR key-path | FROST (m-of-n threshold) | Secret-share setup + multi-round signing protocol. | Very hard on a single signer card; heavy state management. | Probably unnecessary for > 95 % of users in 2025; better served by an external coordinator later

1. First step: SegWit native multisig - P2WSH sortedmulti (Generation 2)

> Wallet - (UR:crypto-psbt fragments)→ Shell camera
>     - MCU verifies descriptor digest + paths
>     - MCU → SE:  sighash digest
>     - SE → MCU:  73-byte DER sig
> Shell display shows: outputs, fee, change ≥ threshold
> Shell - (UR:crypto-psbt)→ Wallet

2. Legacy P2SH multisig (Generation 1) is optional: could support, but it burns 15 % more bytes per spend and very little new infrastructure still requires it

bitgamma, 2025-06-05:
Now can correctly pair with Sparrow, however PSBT parsing needs to be extended to properly support this kind of transactions

phift, 2025-06-06:
The setup in Sparrow works! Haven't tested tx, though.