keycard-tech/keycard-shell #92: Update keycard.tech comparison table

phift, 2025-05-22:
Simplify table on https://keycard.tech/ using our updated comparison: https://docs.google.com/spreadsheets/d/18-AnvJabL-znhucymSOHuUWSJzJax0-oBxq9QXDOb-Y/edit?gid=1099231663#gid=1099231663

guylouis, 2025-05-26:
content of the table is not ok yet
will ping here when the content is good

guylouis, 2025-07-11:
content of the table is ready now, it's here:
https://docs.google.com/spreadsheets/d/18-AnvJabL-znhucymSOHuUWSJzJax0-oBxq9QXDOb-Y/edit?gid=1099231663#gid=1099231663&range=A82

since it's the same format (same number of columns, just a different number of lines) as the existing one, is it really a must to update the figma, or could I make a PR directly on the website source (it that's even possible, just asking)? 

@felicio @jkbktl @marcelines

guylouis, 2025-07-12:
ok so @xAlisher (not sure anyone else I could ping wiht that) whenever you have time could you update the table in https://keycard.tech with the new values of the table like this: 
https://docs.google.com/spreadsheets/d/18-AnvJabL-znhucymSOHuUWSJzJax0-oBxq9QXDOb-Y/edit?gid=1099231663#gid=1099231663&range=A82

xAlisher, 2025-08-06:
@guylouis @jkbktl Updated the table, please review
https://www.figma.com/design/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?node-id=6248-22111&t=2eZlbpuoqilM8nkB-1

closed 2025-09-15 by guylouis