keycard-tech/keycard-shell #112: Add possibility to correct words on Invalid phrase entered screen

phift, 2025-05-29:
### Description
When users enter an invalid phrase during the recovery phrase input process in the backup flow, but doesn't notice until after the 12th word, on the `Invalid phrase entered` screen, they are forced to restart from the beginning. 
This creates frustration, especially when the error is minor (e.g., a single incorrect word). Allowing users to navigate back to a specific word instead of restarting would improve the user experience.

---

#### Key issues:
- Restarting the entire phrase input process for a single error is inefficient and frustrating
- No option exists to edit or revisit a specific word in the phrase
- Users may abandon the process due to repetitive restarts, increasing user friction

---

### Suggested improvements:
- Add navigation controls similar to the current `Import seed phrase` flow https://github.com/keycard-tech/keycard-shell/issues/50 - to allow users to revisit and edit specific words in the phrase

---

### Resources  
https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=0-1

bitgamma, 2025-06-03:
my proposal is to go back to the input of the first word, without deleting the previously inserted words, which will be presented as auto-complete. This way the user reviews each word and corrects the wrong one(s) - just long-pressing OK on correct ones. This consistently requires 12 or 24 long-presses of OK + the input of any correction to finish the flow.

 I think this is better than restarting from the last word, since this would require navigating backward to find the error and then all the way forward again to get to the last word, potentially doubling the number of key presses required

@phift @xAlisher wdyt?

xAlisher, 2025-06-03:
Cool! +1

phift, 2025-06-03:
Sounds good!

closed 2025-06-16 by bitgamma