keycard-tech/keycard-shell #136: Create unified UX to update firmware and token db via cable

xAlisher, 2025-07-10:
Following this discussion it was agreed to combine updating of firmware and token DB in one flow.

https://discord.com/channels/1366754869561790474/1392436342830530653/1392436347117244518

Rsttskyy, 2025-07-12:
Figma 
https://www.figma.com/design/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?node-id=7598-39562&t=7wBBQi0Sogu5kAOM-1

Prototype

Lottie animation 
Demo - https://soft-branding-758712.framer.app/ 

Code: 
`<script
  src="https://unpkg.com/@lottiefiles/dotlottie-wc@0.6.2/dist/dotlottie-wc.js"
  type="module"
></script>

<dotlottie-wc
  src="https://lottie.host/0ab81f68-e20a-446d-9363-b897ff67bd50/cLYIbtLguC.lottie"
  style="width: 590px;height: 344px"
  speed="1.6"
  autoplay
  loop
></dotlottie-wc>`

Rsttskyy, 2025-07-12:
Ultra wide screen layout behavior 
https://www.figma.com/design/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?node-id=7833-16867&t=i0W6WuJrkITjxoUF-1

Demo (resize window to see behavior) 
https://soft-branding-758712.framer.app