keycard-tech/keycard-shell #138: Update visual style of standalone updater

xAlisher, 2025-07-10:
Update visual style and assets of standalone updater to match with Keycard Shell website visual style. 
Link to current version: https://github.com/choppu/kshell-updater

Rsttskyy, 2025-07-28:
https://www.figma.com/design/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?node-id=6088-46702&t=uQ8mPUJbtZCPFsW3-1

xAlisher, 2025-07-28:
Review UI details with @Rsttskyy

choppu, 2025-10-04:
Yes, this one can be closed. This one is already implemented.

closed 2025-10-04 by guylouis

Rsttskyy, 2025-10-05:
Icon updated
https://www.figma.com/design/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?node-id=6987-6811&t=tBQqcyfvaxFwOYsE-1 

Changes: 
- Clear case added
- orange bg changed to black

Reason: orange not works visually good combined with clear case.