keycard-tech/keycard-shell #139: Have a .stl project of Shell on printable.com

guylouis, 2025-07-12:
The current files (steps) we have for Shell have been designed and optimized for plastic injection in our steel mold. If we want diy home builders to print a casing and use it with their electronics, we can help them by 
- exporting the designs to a .stl files
- check/experiment/modify them to make sure the assembly works 
- upload them on printable.com

xAlisher, 2025-07-14:
I'm in process of communication with Printables

xAlisher, 2025-08-11:
Printables do not reply.

Considering alternatives.

Thingiverse – The original big repository for 3D printing files, still widely used.

Cults3D – Popular with designers selling or freely sharing STL and other files.

Thangs – 3D model search and hosting with collaboration tools.

Libre3D – Fully open-source repository for 3D printable models.