keycard-tech/keycard-shell #140: Create web-site menu for Shell owners

xAlisher, 2025-07-12:
Details

https://discord.com/channels/1366754869561790474/1366754869561790477/1393200542666461275

Rsttskyy, 2025-07-12:
Figma
https://www.figma.com/design/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?node-id=7797-71401&t=7wBBQi0Sogu5kAOM-1

Prototype
https://www.figma.com/proto/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?page-id=7797%3A71400&node-id=7797-71402&viewport=628%2C568%2C0.49&t=BE1I3lQkKkqm5Dsk-1&scaling=min-zoom&content-scaling=fixed&starting-point-node-id=7797%3A71402

Rsttskyy, 2025-07-17:
added mobile view

Figma 
https://www.figma.com/design/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?node-id=7983-34559&t=Vgk6no6cj3hTGI9z-1

Prototype 
https://www.figma.com/proto/RSszWiHjK6os09PEDmJ0gR/Website---Keycard?page-id=612%3A16059&node-id=1584-16480&viewport=-5440%2C16010%2C1.09&t=Qg8gcBPpBjszKEEd-1&scaling=min-zoom&content-scaling=fixed&starting-point-node-id=1584%3A16480