keycard-tech/keycard-shell #144: slip39

guylouis, 2025-07-24:
slip39 is a seed backup system based on shamir secret sharing scheme, and is an improvement to 12 or 24 words bip39 scheme. It is a flexible system that solves bip39 single point of failure issue by sharing the secret in several word lists. 

slip39 is getting traction in our industry with support from 
- hardware wallets: Trezor (slip39 being default mode) and Keystone
- Electrum, Sparrow Wallet, BlueWallet, Rabby, NuFi

https://slip39.com/
https://content.trezor.io/slip39

Integrating SLIP39 in Shell would make sense, with the following steps: 
1/ mono-share generation and import
2/ multi-share import
3/ multi-share generation

Note: as shown in the tool presented in https://slip39.com/ multishare can actually be done on n-ou-of-m groups, and within _each_ group with x-out-of-y cards. We consider this would be too complex of a UI for a hww, and just like Trezor did, would stick to one group, so that the scheme we implement is a _single_  x-out-of-y scheme.

xAlisher, 2025-07-28:
Generating a new key pair via slip39 https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=5472-72002&t=eDyX5R9rPkm170b8-1

xAlisher, 2025-07-28:
Importing SLIP39 https://www.figma.com/design/I4NT4JGHDEVSQ2yo5Z4mju/Shell-OS?node-id=5493-77424&t=tcmZ9vQxYsRMp45E-1

bitgamma, 2025-07-28:
development steps

- [ ] implement RS1024 checksum
- [ ] implement Shamir interpolation
- [ ] implement secret generation, combination, split
- [ ] implement secret encryption/decryption
- [ ] implement SLIP39 encoding of secret
- [ ] add key import UI flow for SLIP39
- [ ] add key generation UI flow for SLIP39

xAlisher, 2025-08-11:
@romanzac Let's test it!

romanzac, 2025-08-22:
> [@romanzac](https://github.com/romanzac) Let's test it!

Tests finished with no issue found on the Shell's side.
Results available at https://www.notion.so/2548f96fb65c8036961ef84e3cbc71cc?v=2548f96fb65c80b0b2ed000c09db9e3a&source=copy_link

closed 2025-08-22 by bitgamma