keycard-tech/keycard-shell #151: design and print sample of the regulatory leaflet

guylouis, 2025-07-30:
For regulatory reasons in europe, usa, and canada we need to print a leaflet that we will put in Keycard shell box. 

This leaflet includes 
- use and care recommendations for the product
- warnings for the battery
- european regulatory declaration
- usa regulatory declaration 
- canada regulatory declaration 

Here's a draft for the content of the leaflet https://docs.google.com/document/d/1JTA68iDlfdyYfmAgcMc36htxLo5xAx9b8aHBHjhYCi8/edit?usp=sharing

guylouis, 2025-08-20:
These suppliers tend to ask us to provide full detail about the formats we want, so @Francesca-G may you put the 6 pages just like the picture above on one sheet
- printed only on the front (no back printing)
- same folding idea (we can use the picture above to illustrate the folding we want)
- as for the size let's consider a size that makes it compatible with putting it on the top compartment (but I think we will put it on the bottom compartment), it leaves us some flexbility. 

Thanks!

Francesca-G, 2025-08-22:
- Size when the leaflet is close: 65x100mm (same as Quick Start Guide card) 
- Total size of the leaflet (6 pages): 195x200mm

[Regulatory_Leaflet_195x200.pdf](https://github.com/user-attachments/files/21935236/Regulatory_Leaflet_195x200.pdf)

closed 2025-09-08 by guylouis